easytarget/esp32-cam-webserver #189: esp32cam programming problem

Patsjemoe, 2021-11-27:
Hi,
My newest esp32cam cannot be programmed , although I do exactly the same as with my other esp32cam module.
I noticed that my latest module has a different esp32 module on board, not from AI Thinker ...but unknown, do not recognise the logo.

I attached a picture of both : 
[image]
[image]

Is this a fake device ?
Do I need to select another module then AI Thinker esp32cam ?
I alsready changed to programming module...not better.
any suggestion ?
Regards
Ludo

Kallemanni, 2021-11-27:
Hey there, I also had different problems with my boards and I have the exact same as the one that you got.
For me it helped to update the ESP32 Core for Arduino IDE to version 1.0.6. ... maybe it will work for you also?
In addition you may try the ESP32 Dev Module and enable the PSRAM.
Regards ...

Patsjemoe, 2021-11-28:
Hi Kallemanni,
Thank you for your quick suggestions.
Have already version 1.0.6....
Tried ESP32 Dev. Module, but also not working.
It just does not connect to the module. Tried different settings, nothing
works.
If anyone has other suggestions, they are also welcome.
Best regards
Ludo

Patsjemoe, 2021-11-29:
Hi,
SOLVED 
Needed settings according attachmen
[image]
t :

closed 2021-11-29 by Patsjemoe