easytarget/esp32-cam-webserver #190: master branch - Camera sensor failed to initialize: allocating framebuffer failed

danilsh, 2021-11-28:
I have this kit: [https://aliexpress.ru/item/1005003334474586.html](https://aliexpress.ru/item/1005003334474586.html)

I tried a default example CameraWebServer from Arduino IDE 1.8.16 ESP32 package 1.0.6, selecting AI Thinker Board. All works just fine.

When I try to flash your firmware from master branch, it does not works. Server displays error page. Here is full log:

`
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled

====
esp32-cam-webserver: ESP32 camera server
Code Built: Nov 28 2021 @ 20:56:35
Base Release: 4.0.beta2
[E][camera.c:267] camera_fb_init(): Allocating 375 KB frame buffer Failed
[E][camera.c:1281] camera_init(): Failed to allocate frame buffer
E (1115) gpio: gpio_isr_handler_remove(398): GPIO isr service is not installed, call gpio_install_isr_service() first
[E][camera.c:1406] esp_camera_init(): Camera init failed with error 0x101

CRITICAL FAILURE: Camera sensor failed to initialise.

A full (hard, power off/on) reboot will probably be needed to recover from this.
Meanwhile; this unit will reboot in 1 minute since these errors sometime clear automatically
Lamp: 0%, pwm = 0
Starting WiFi
[W][WiFiGeneric.cpp:637] setSleep(): STA has not been started
Known external SSIDs:  'CHTON'
MAC address: 4C:EB:D6:7C:CB:20
Scanning local Wifi Networks
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 1 - SCAN_DONE
1 networks found
  1 : [60:45:CB:C9:3E:90] CHTON (-78)  -  Known!
Connecting to Wifi Network 0: [60:45:CB:C9:3E:90] CHTON 
........[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:419] _eventCallback(): STA IP: 192.168.1.248, MASK: 255.255.255.0, GW: 192.168.1.1
.Client connection succeeded
IP address: 192.168.1.248
Netmask   : 255.255.255.0
Gateway   : 192.168.1.1
Setting httpURL
Setting up OTA

No OTA password has been set! (insecure)

[I][ArduinoOTA.cpp:141] begin(): OTA server at: ESP32 camera server.local:3232
Time functions disabled
Starting web server on port: '80'
Starting stream server on port: '81'

Camera unavailable due to initialisation errors.

This is the 4.0 alpha
 - Face detection has been removed!
`

skyblueplus, 2021-12-20:
I'm having the same issue.

easytarget, 2021-12-20:
What is the output if you send `d` in the serial console to generate a dump
(or go to `/dump` in the web interface)

skyblueplus, 2021-12-20:
`Name: ESP32 camera server
Firmware: Dec 20 2021 @ 10:13:18 (base: 4.0.beta2)
Sketch Size: 1040784 (total: 1966080, 52.0% used)
MD5: 5ff3fc791e148c298b8a6fcf61a25829
ESP sdk: v3.2.3
WiFi Mode: AccessPoint with captive portal
WiFi SSID: ESP32-CAM-CONNECT
WiFi IP address: 192.168.4.1
WiFi Http port: 80, Stream port: 81
WiFi MAC: XX
System up: 0:00:00:17 (d:h:m:s)
Active streams: 0, Previous streams: 0, Images captured: 0
Freq: 240 MHz
Heap: 294536, free: 185124, min free: 168324, max block: 113792
Psram: 2097100, free: 2097100, min free: 2097100, max block: 2097100
Spiffs: 0, used: 0
Preferences file: 
/esp32cam-preferences.json not found, nothing to dump.`

Seems something's wrong with spiffs?

easytarget, 2021-12-20:
Lack of Spiffs wont cause this issue.
You appear to be running a very old release of the expressif IDE core; v 1.0.4 or earlier. 
Please update to the latest IDE release (2.0.1) and be sure to set the Board as shown in the readme.

skyblueplus, 2021-12-20:
After updating to 2.0.1 release, camera initialization still fails:

E (562) cam_hal: cam_dma_config(280): frame buffer malloc failed
E (563) cam_hal: cam_config(364): cam_dma_config failed
E (563) camera: Camera config failed with error 0xffffffff

Or after some rebooting, with this error:

Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)

Dump result:

Name: ESP32 camera server
Firmware: Dec 20 2021 @ 22:10:10 (base: 4.0.beta2)
Sketch Size: 1003376 (total: 1966080, 51.0% used)
MD5: 6de12343b4510327cfe821a4f1d986aa
ESP sdk: v4.4-dev-3569-g6a7d83af19-dirty
WiFi Mode: AccessPoint with captive portal
WiFi SSID: ESP32-CAM-CONNECT
WiFi IP address: 192.168.4.1
WiFi Http port: 80, Stream port: 81
WiFi MAC: XX
System up: 0:00:00:26 (d:h:m:s)
Active streams: 0, Previous streams: 0, Images captured: 0
Freq: 240 MHz
Heap: 282236, free: 198420, min free: 194224, max block: 65524
Psram: 2094315, free: 2085863, min free: 2080547, max block: 1048564
Spiffs: 0, used: 0
Preferences file: 
/esp32cam-preferences.json not found, nothing to dump.
A critical error has occurred when initialising Camera Hardware, see startup megssages

I also tried connecting to other power source, erasing flash and commenting the led pins from camera_pins.h, which didn't help.

easytarget, 2021-12-21:
Hummm.. I'm wondering if this is connected to #175 

Could you please add:
```delay(1000);```
to `esp32-cam-webserver.ino` on line [639](https://github.com/easytarget/esp32-cam-webserver/blob/master/esp32-cam-webserver.ino#L639) and re-upload + test.

This will add a 1 second delay after the camera module is told to init but before starting to initialise the spiffs partition etc.

skyblueplus, 2021-12-21:
Unfortunately, the same error occurs. Tested with 1000 and 2000 delay.

easytarget, 2021-12-31:
OkiDoki; I see two things I don't understand in your config:
```
Psram: 2094315, free: 2085863, min free: 2080547, max block: 1048564
Spiffs: 0, used: 0
```

Here is what I see on my AI-THINKER boards:
```
Psram: 4191419, free: 3415203, min free: 3399419, max block: 3407860
Spiffs: 173441, used: 753
```

The low PSRAM and lack of SPIFFS says that something is either wrong with your hardware, or wrong with your config.
Are you using the Arduino IDE or some other tool (PlatformIO/esp-ide) for programming?
Why does the ESP-SDK show as `ESP sdk: v4.4-dev-3569-g6a7d83af19-dirty`? I do not see the `-dirty` tag in my environment, this indicates the SDK has been altered since it was installed; could these changes affect the board parameters you use?

Anyway.. I do have something concrete you can try:
Update to the new beta3 release and edit myconfig. 
- Uncomment and change the default xclk (camera bus clock) setting to 2MHz on the very last line:
  ```#define XCLK_FREQ_HZ 2000000;```
- The default is 16MHz, some boards can go over 20MHz, but other recent clone boards have issues with this and need much lower speeds.
- I have a recent board that will fail to start the camera unless the speed is below 8MHz, and shows image corruption in high resolution modes until I lower the speed to 2MHz
I'm curious to see if this helps for your errors, you can double-check the xclk speed in the dump output now too.

LordGaav, 2022-01-08:
I seem to be running into the same issue with the latest master branch:

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1324
ho 0 tail 12 room 4
load:0x40078000,len:13508
load:0x40080400,len:3604
entry 0x400805f0

====
esp32-cam-webserver: ESP32 camera server
Code Built: Jan  8 2022 @ 22:14:18
Base Release: 4.0.beta3
Camera init succeeded
OV2640 camera module detected
E (1332) SPIFFS: mount failed, -10025
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1324
ho 0 tail 12 room 4
load:0x40078000,len:13508
load:0x40080400,len:3604
entry 0x400805f0

====
esp32-cam-webserver: ESP32 camera server
Code Built: Jan  8 2022 @ 22:14:18
Base Release: 4.0.beta3
E (7919) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)

CRITICAL FAILURE: Camera sensor failed to initialise.

A full (hard, power off/on) reboot will probably be needed to recover from this.
Meanwhile; this unit will reboot in 1 minute since these errors sometime clear automatically
Lamp: 0%, pwm = 0
Starting WiFi
Known external SSIDs:  XXX
MAC address: 24:0A:C4:ED:EC:3C
Scanning local Wifi Networks
Connecting to Wifi Network 0: XXX
..........Client connection succeeded
...
Setting httpURL
Setting up OTA

No OTA password has been set! (insecure)

Time: 22:17:19, Saturday, January 08 2022
NTP Server: nl.pool.ntp.org, GMT Offset: 3600(s), DST Offset: 3600(s)
Starting web server on port: '80'
Starting stream server on port: '81'

Camera unavailable due to initialisation errors.

This is the 4.0 alpha
 - Face detection has been removed!

```

I tried without configuring the external clock, 3 Mhz and 20 Mhz, no difference. What seems curious is that early on there is a line "Camera init succeeded", but later on it fails?

easytarget, 2022-03-08:
Rolling this and all related into a single issue: https://github.com/easytarget/esp32-cam-webserver/issues/214

@LordGaav : It really sounds like your board has some sort of hardware issue with the camera bus clock or memory

closed 2022-03-08 by easytarget