easytarget/esp32-cam-webserver #191: v3.4 app_httpd.cpp fatal error "fd_forward.h" no such file

ericfont, 2021-12-03:
Tried your v3.4 branch and got:

```
Multiple libraries were found for "WiFi.h"
app_httpd.cpp:70:10: fatal error: fd_forward.h: No such file or directory
 Used: /home/e/.arduino15/packages/esp32/hardware/esp32/2.0.1/libraries/WiFi
 #include "fd_forward.h"
 Not used: /home/e/arduino-1.8.16/libraries/WiFi
          ^~~~~~~~~~~~~~
compilation terminated.
exit status 1
fd_forward.h: No such file or directory
```

Though your latest master (0ef997e21bfe597ad25d366f9acef99891dd0d09) compiles fine.

easytarget, 2021-12-16:
And OOPS:
https://github.com/espressif/arduino-esp32/issues/5878#issuecomment-966756467
FR is removed from the ArduinoIDE from 2.0.1+
- Not noted in release notes
- not documented properl;y elsewhere either.

I'm going to ditch supporting the 3.x branch now, I think. Too much effort to keep it working.
- I'll wrap it up with a note that you can downgrade to esp-ide 2.0.0 if you really want to see Face Recognition in action.

easytarget, 2021-12-31:
See #196 
- The suggested solution there also covers fixing this.

danielgeiser, 2022-01-20:
For me the suggested solution didn't work.

easytarget, 2022-03-08:
The 3.x branch (with face recognition) now specifies ESP Arduino IDE core 1.0.6 as it's only supported core. 
The 4.x branch has no face recognition and is not affected.

If you see this error it is because you are trying to compile the 3.x version of the cam webserver with a 2.x version of the ESP Arduino Core; but they are not compatible:
Please see the 3.x readme; specifically here:
https://github.com/easytarget/esp32-cam-webserver/blob/3.x/README.md#downgrade-the-esp-arduino-core-to-v106

closed 2022-03-08 by easytarget