easytarget/esp32-cam-webserver #192: Pan & Tilt servo motor control

antonblaise, 2021-12-10:
I would like to install the ESP32 cams onto my pan-tilt  (x and y axes) servo motor brackets, as I wish to be able to control where the camera is pointed at. There are lots of arduino codes out there that does this, but I truly wish that this feature can be added here, as this is by far my most favourite ESP32 cam webserver of all.

easytarget, 2021-12-15:
Sorry, but I wont be doing any work on this. 
While nice to have I do not have the time or motivation to implement anything like this.

I will, however, leave this open since it is a good enhancement idea, if anybody else wants to work on it I'd be happy to accommodate any well-written PRs

ZalgoSoft, 2022-06-19:
glad to know you accepted my ptz code.
Recently enchanced  my code for smooth movement.