easytarget/esp32-cam-webserver #196: PSRAM now a requirement?

easytarget, 2021-12-17:
Looking at recent errors I see a common thread:
- camera modules failing to init
- `NO PSRAM FOUND`
eg:
- #182
- #176

Test with disabling psram and see whether it is now a hard limit with the latest esp cores.
- might still be just co-incidence and flaky modules and/or low voltages

tobiasisenberg, 2021-12-31:
Just as a note: it was indeed a configuration issue for [the now closed Issue 182](https://github.com/easytarget/esp32-cam-webserver/issues/182) that I had forgotten to enable ```PSRAM```; with it enabled the code works fine. My hunch would be that this is also true for [issue 176](https://github.com/easytarget/esp32-cam-webserver/issues/176).

easytarget, 2021-12-31:
This seems pretty much confirmed: Startup console log for my new module when PSRAM is disabled:
```
====
esp32-cam-webserver: AquaCam
Code Built: Dec 31 2021 @ 12:52:13
Base Release: 4.0.beta3
E (132) cam_hal: cam_dma_config(280): frame buffer malloc failed
E (132) cam_hal: cam_config(364): cam_dma_config failed
E (133) camera: Camera config failed with error 0xffffffff

CRITICAL FAILURE: Camera sensor failed to initialise.
```
then, after all the webserver init etc..
```
Camera unavailable due to initialisation errors.

No PSRAM found.
Please check the board config for your module.
```
Re enabling PSRAM and rebuilding results in:
```
====
esp32-cam-webserver: AquaCam
Code Built: Dec 31 2021 @ 13:11:51
Base Release: 4.0.beta3
Camera init succeeded
OV2640 camera module detected
```
and a dump shows:
```
Firmware: Dec 31 2021 @ 13:11:51 (base: 4.0.beta3)
Sketch Size: 998176 (total: 1966080, 50.0% used)
MD5: c99c0a949c5e5cddfb02ee81f28c90ff
ESP sdk: v4.4-beta1-189-ga79dc75f0a
...
Psram: 4191479, free: 3415779, min free: 3413979, max block: 3407860
```

easytarget, 2021-12-31:
Plan:
- Move PSRAM check to the head of the setup, and hard-fail early with a 'No PSRAM, no CAM' message
  - Point people to the README for mis-configurers, and the v3 branch for those without PSRAM.
- Update 3.x with instructions for how to downgrade to esp-core 1.0.6
  - or maybe esp-core 2.0.0 if it works, I need to test.
  - platformio and travis configs need to be rolled back to match.
- Promote 3.x as the build to use if you want to look at Face Recognition, or have no PSRAM
- Add caveats about downgrading, and warnings that the esp-core needs to be upgraded back to the latest version after programming

easytarget, 2022-03-08:
Upcoming 4.0 release has a more prominent warning about not having PSRAM, given much earlier in the startup. The web `/dump` page already hasa prominent (red) note about it too.
I'm not going to *fail* the sketch for this, it will still start and give a web UI, just with poor quality video.

commit 6d825277 (2022-03-08): Fail if no PSRAM, fix #196
closed 2022-03-08 by easytarget