easytarget/esp32-cam-webserver #200: New board to add - minor changes

rdragonrydr, 2021-12-24:
My personal board is the Arducam Uno. Here's the Boards.txt: https://github.com/ArduCAM/ArduCAM_ESP32S_UNO/blob/master/variants/ArduCAM_ESP32S_UNO/pins_arduino.h

I got it working with the only change I found (YMMV) being to set the notification LED to pin 2.

easytarget, 2021-12-24:
Thankyou; I'll try to fold this into a little hotfix I'm doing.

rdragonrydr, 2021-12-24:
I was doing some more reading and I noticed there's a setting for whether the LED is active HIGH or LOW. In my case, the notification LED is normally on and blinks off when there is activity, and I haven't modified the camera configs for that.

(Speaking of which, I forgot to mention, but it is using CAMERA_MODEL_AI_THINKER.)

I'm no longer sure this was the intended function, but I like it. If that isn't the normal operation, would it be possible to add an InvertLED setting to the myConfig?

easytarget, 2021-12-31:
I just rolled the new board definition out as part of beta3: 4b6cd9a56ef75
I changed the notification invert status and pin to the arduino default, and removed the illumination lamp pin definition too since this board doesn't have that as far as I can see.

One of my concerns is that the config file is getting bloated, so I'm not going to expose a specific option for this, it's not a bad suggestion but I think it's easy enough for people to invert via `camera_pins.h` as needed.

closed 2021-12-31 by easytarget

rdragonrydr, 2021-12-31:
A minor quibble - I actually manually added a light to mine since, yes, it didn't have one. I used the default for the AI-Thinker, which apparently is pin 4?

Also, if removing the config still leaves the light controls in the app, it might be better to tell the users what pin that is and leave it enabled anyway so they can add their own as I did.

rdragonrydr, 2021-12-31:
Also, the selection list in MyConfig does not have the new board. I've added it to mine, but you may have wanted to add that and forgot?

easytarget, 2021-12-31:
I dont actually provide `myconfig.h `as part of the repo, only the sample template.
A better note on updating is probably needed, noting that config changes need to be merged

rdragonrydr, 2021-12-31:
...Yeah, that's on me. I should have realized that, as I knew the file wasn't tracked in git. Oops!