easytarget/esp32-cam-webserver #201: Does this have MDNS?

rdragonrydr, 2021-12-24:
I remember seeing a mention of MDNS made in the platformio.ini file and I'd like to be able to address my board by something other than IP address.

However, I can't seem to find its MDNS address and I can't seem to find it in the code, either. Is this not using MDNS or is it my network being weird?

easytarget, 2021-12-24:
No; I'm afraid it doesn't at present. But this is a good suggestion, so I'm going to keep it open.
- The mention in platformio.ini is unrelated; it just specifies how the uploader finds the OTA target.

A 5-minute google says to me that it wouldn't be too difficult to add:
- https://github.com/espressif/arduino-esp32/blob/master/libraries/ESPmDNS/examples/mDNS_Web_Server/mDNS_Web_Server.ino
- https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/mdns.html

However; this has to go on the back-burner since I have some bigger fish to fry first.

If anybody else wants to have a crack at it.. I'm all PR's :-)

rdragonrydr, 2021-12-24:
Would the literal example code work here (admittedly with the myConfig stuff added in to support name changes)? Or would I need to add other protocols/services? I think it'd need one for OTA, the camera stream, and the web server.

I remember seeing the ports for the server and stream listed, and of course the server has a known protocol too. I am not sure how to add OTA at all, or what to say the stream's protocol is.

I've also never made a PR before, so this might be a "fun" experience for me... :D

easytarget, 2021-12-27:
[MDNS](https://en.wikipedia.org/wiki/Multicast_DNS) is just a way for devices like to cam to announce their name and services+address on the network so they can be found by other machines, think of it as a private micro DNS server..

I think that all you need to do is:
- [include](https://github.com/espressif/arduino-esp32/blob/caef4006af491130136b219c1205bdcf8f08bf2b/libraries/ESPmDNS/examples/mDNS_Web_Server/mDNS_Web_Server.ino#L20) the mDNS headers in the sketch.
- once the network is connected:
  - If OTA is Enabled and starts properly it will start the mDNS service for you ([see this](https://www.esp8266.com/viewtopic.php?p=49823#p49823)) 
  - If OTA is Disabled or fails to start you need to  [run](https://github.com/espressif/arduino-esp32/blob/master/libraries/ESPmDNS/examples/mDNS_Web_Server/mDNS_Web_Server.ino#L48) the MDNS service.
    [*do not* go into an infinite loop if it errors.. just report the error and continue]

If it started OK [add the http service](https://github.com/espressif/arduino-esp32/blob/master/libraries/ESPmDNS/examples/mDNS_Web_Server/mDNS_Web_Server.ino#L66) to it; the OTA service will already have been added if necessary by OTA setup. I don't think you need to announce the stream port separately.

Good luck! there is lots of info on PR submitting out there; and I have a guide for this project:
[CONTRIBUTING.md](https://github.com/easytarget/esp32-cam-webserver/blob/master/CONTRIBUTING.md)

Finally; when looking at that example note that the [main() loop](https://github.com/espressif/arduino-esp32/blob/master/libraries/ESPmDNS/examples/mDNS_Web_Server/mDNS_Web_Server.ino#L69) has nothing whatsoever  to do with mDNS, it's just a simple web server demo that acts as the target for the MDNS record created on line 66.

rdragonrydr, 2022-03-07:
I have a merge ready to go, but I'm confused by a note in the CONTRIBUTING.MD guide.
[image]
Are these the correct settings? I think the line "Issue a pull request against a branch of the same name in the main repo." means that I need to change the branch in the left box away from Master, right?

rdragonrydr, 2022-03-07:
Also, I apologize preemptively, but it appears that some of the line endings have been normalized. I turned off all my code styling plugins, but it appears that my IDE itself did something.

easytarget, 2022-03-09:
Thankyou for working on this!

Dont worry about the line endings; IDE's and editors are annoying that way; I'm used to dealing with it ;-)

The CONTRIBUTING guide is wrong, I'll update that. 

Please submit this against the default 'master' branch; I'll then rebase the changes into my 4-0-rc1 branch.

easytarget, 2022-03-09:
alternatively.. since I just ran a line-ending whitespace stripper across the 4-0-rc1 branch:
update your fork (there is a button for that in github)
checkout the 4-0-rc1 branch and redo your changes on that
submit a PR against my 4-0-rc1 branch.

easytarget, 2022-03-09:
I just deleted a comment; because it made a bad suggestion, on reflection
(I'll be pushing to my 4-0-rc1 branch at will, so this would force you into rebase hell..)
> alternatively.. since I just ran a line-ending whitespace stripper across the 4-0-rc1 branch:
> update your fork (there is a button for that in github)
> checkout the 4-0-rc1 branch and redo your changes on that
> submit a PR against my 4-0-rc1 branch.

rdragonrydr, 2022-03-09:
I've submitted a PR against Master as requested. If you need me to do any changes please let me know.

easytarget, 2022-03-23:
I forgot to mark this as resolved, the changes are in the 4.0 release.
Thanks again for the suggestion and PR

closed 2022-03-23 by easytarget