easytarget/esp32-cam-webserver #202: Streaming server not working for OV2640

SomeDevWeb, 2021-12-25:
Hi good people. 
First of all Merry Christmas and Happy new year! :)

Secondly, I am testing a Waveshare Esp32 One board with an OV2640 cam.
Everything went OK, I read your doc and tweaked a few things and all actions behave great, except for the streaming server that almost never works. It only fired once or twice.
**Note1**: the cam definition that worked is 'CAMERA_MODEL_ESP_EYE'
**Note2**: the streaming works with the default espressif webcam example sketch. 

Does anyone have experienced this or have any idea if I'm doing anything wrong? 
Thanks a lot!

founfabug, 2021-12-25:
Just a quick suggestion. I had some troubles when testing and discovered that one of the chrome-based browsers would keep the stream open, so subsequent attachments would not be able to access the stream. Had me tricked for a while as single images worked fine. A reset of ESP would clear it, but on occasion, the browser's agressive precaching would attach again. Especially tricky when testing from phone and pc.

easytarget, 2021-12-26:
This is a well know issue, keeps catching people out. See #51

closed 2021-12-26 by easytarget

SomeDevWeb, 2021-12-26:
Hi. 
I was just gonna reply. This issue closing was surprisingly quick... :)
Anyway, sorry I think you're right up to a point, but I have to disagree in the end. 
Regardless of the reason that causes the issue, I strongly believe that this is one of the most important features of the software and if one can't see the camera output it in 99 percent of the trials (as it's my case too) then there's no point really to have that option there...
I actually think it doesn't matter too much that the stream can't be multi-user for time being, but the bottom line is there's gotta be a way to at least detect if the stream is already used and it should be disconnect from that, so the current access can be assured. 
I will do some research too and look through the code a bit, even though I am mostly a web dev. But I hope I can see something and I'll send any feed-back.
Thanks again for this software. Well done!

easytarget, 2021-12-26:
> one can't see the camera output it in 99 percent of the trials (as it's my case too) 

You got encountered this when you tried to run from multiple clients at the same time, not realising that is not supported. That's not the typical use-case, and most people do not run afoul of this as quickly as your average web-developer :stuck_out_tongue: 
- It really should be better documented.
- The official example this is based off has exactly the same issue, the solution is to properly implement multi-client streaming.
  But that's a really big task. Beyond the scope of work I have time to do.
- A change where a second connection to the stream closes any existing stream isn't a great solution, stream hijacking shouldn't be a feature. It's first-come, first served.
- It might be better to send a 403 / 503 response to subsequent stream requests, at least the browser will fail properly with that.

SomeDevWeb, 2021-12-27:
I understand and agree with your points, so just earlier I tried the original example again.
I'm just saying this for the benefit of your project, not at all to contradict you, but I'm afraid that example is a lot more reliable, in the sense yes, it has the same behaviour to only allow one stream at a time, but when I try from the phone and PC, they cancel each other, but every time the stream shows on at least one of them, whereas with this script the same test only shows the stream image maybe once from tens of trials... :(
Sorry, it could be something I AM DOING WRONG, but I tried to follow your style of setting things in the myconfig.h file.
One note is that I tried the original example without the AP mode, I don't know if that makes any difference.

Either way, I think for time being I will continue for a while with that example, especially since the few things I want to implement, as it happens, are not in this script. 
Again, this is a development suggestion for this project, the things I need are:
- A "Stop/Shutdown" button that doesn't actually shut it down, but will put the board in the deepest sleep mode. Here I wonder if the code can still start it from the deep sleep mode... will have to research... :)
- A dynamic WIFI setup. I know this has the AP mode, which is great, but that is slightly different
- An option to save the images on the SD card. I believe now it saves them in the flash area, no?

Anyway, your efforts are great and it's amazing you are offering this as FOSS. 
I'll try to come back with some updates if/when I have...