easytarget/esp32-cam-webserver #204: Issue with LAMP_DEFAULT

rdragonrydr, 2021-12-31:
I've added back the config value for my manually-added illumination LED. It works via the server's light control slider. However, the default value setting (#define LAMP_DEFAULT 3) is not making it light up. I was trying to have the LED light up dimly when it was on so that (barring me manually changing it) I'd know if the camera is running as I'm operating it off of batteries.

The default lamp value is not working, and the LED is staying off. Again, I can set it with the server, and I see the default value getting applied on startup in the serial logs, but it doesn't actually illuminate. I also tried the old AITHINKER config that had the wrong notification LED but that I know worked with the lamp pin I was using. It still doesn't work.

I think it might be an initialization order issue, as I've run out of things I know to test for as to why it might not be working.

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12776
load:0x40080400,len:3032
entry 0x400805e4

====
esp32-cam-webserver: ESP32 camera server
Code Built: Dec 31 2021 @ 13:27:28
Base Release: 4.0.beta3
Camera init succeeded
OV2640 camera module detected
Internal filesystem contents
Listing SPIFFS directory: /
Preference file /esp32cam-preferences.json not found; using system defaults.
Lamp: 3%, pwm = 7
Starting WiFi
Known external SSIDs:  '----'
MAC address: -------
Scanning local Wifi Networks
4 networks found
  1 : ------
  2 : ------
  3 : ------  -  Known!
  4 : ------
Connecting to Wifi Network 1: ------
..........Client connection succeeded
IP address: ------
Netmask   : ------
Gateway   : ------
Setting httpURL
Setting up OTA

No OTA password has been set! (insecure)

Time: 23:33:27, Friday, December 31 2021
NTP Server: 0.us.pool.ntp.org, GMT Offset: 18000(s), DST Offset: 3600(s)
Starting web server on port: '80'
Starting stream server on port: '81'

Camera Ready!
Use '-----/' to connect
Stream viewer available at 'http://-------/view'
Raw stream URL is 'http://------/'
Camera debug data is disabled (send 'd' for status dump, or any other char to enable debug)
E (14360) cam_hal: EV-EOF-OVF
E (14360) cam_hal: EV-VSYNC-OVF
E (14676) cam_hal: EV-EOF-OVF
E (14726) cam_hal: EV-EOF-OVF
E (14726) cam_hal: EV-VSYNC-OVF
E (14855) cam_hal: EV-EOF-OVF
E (15186) cam_hal: EV-EOF-OVF

This is the 4.0 alpha
 - Face detection has been removed!
 ```

kawht, 2022-01-08:
I am having the same problem.

easytarget, 2022-03-09:
Tested; confirmed. Hummm.

easytarget, 2022-03-09:
Ok; I have changed the order for this; it works properly on my test rig.
The lamp is now started near the end of the startup, after camera and wifi have finished doing their thing.
If you have a value saved in the preferences file it will override the default value from myconfig.

commit cb78162e (2022-03-09): lamp default state fix #204
closed 2022-03-09 by easytarget