easytarget/esp32-cam-webserver #205: CRITICAL FAILURE: Camera sensor failed to initialise. A full (hard, power off/on) reboot will probably be needed to recover from this. Meanwhile; this unit will reboot in 1 minute since these errors sometime clear automatically

Paarth-go, 2022-01-02:
Hello everyone, I have ESP-32 Cam model Ai-THINKER (clone) with esp32-cam-webserver code uploaded. Till yesterday it was working fine but when i was moving the camera it suddenly stopped working and gave me this error:.

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6360
entry 0x400806b4

====
esp32-cam-webserver: ESP32 camera server
Code Built: Jan  2 2022 @ 21:18:27
Base Release: 4.0.beta2
[E][camera.c:1113] camera_probe(): Detected camera not supported.
[E][camera.c:1379] esp_camera_init(): Camera probe failed with error 0x20004

CRITICAL FAILURE: Camera sensor failed to initialise.

A full (hard, power off/on) reboot will probably be needed to recover from this.
Meanwhile; this unit will reboot in 1 minute since these errors sometime clear automatically
Lamp: 0%, pwm = 0
Starting WiFi
Known external SSIDs: None
MAC address: 94:B9:7E:E6:0D:1C
AccessPoint mode selected in config
Setting up AccessPoint
  SSID     : ESP32-CAM-CONNECT
  Password : InsecurePassword
IP address: 192.168.4.1
Setting httpURL
Starting Captive Portal
Setting up OTA

No OTA password has been set! (insecure)

Time functions disabled
Starting web server on port: '80'
Starting stream server on port: '81'

Camera unavailable due to initialisation errors.

This is the 4.0 alpha
 - Face detection has been removed!

[image]

founfabug, 2022-01-03:
As nothing else changed, you can try re-seating the camera connector. It may have come out of alignment with the move. Videos available online, just be gentle.

Paarth-go, 2022-01-03:
I removed the camera sensor 10 times but no change, then I reuploaded the but same

bolulu, 2022-01-05:
I got the same picture, with Ai-thinker CAM
[image]

easytarget, 2022-01-05:
### I'm moving this to a discussion since It is a good issue to illustrate this common problem

There are numerous ways these failures can appear. 
- With a bit of googling you will find that initialisation errors are very, very common with these modules.
- There are both hardware, and software related reasons for this.
- None of them are particularly related to this sketch (which uses the same initialisation+setup as the official example)

### Typical causes:
- Low power/voltage to the unit
  - this is the number 1 cause, and can strike randomly
  - cheap clones are extremely sensitive to this
  - leaving the unit unplugged for a while can help. I think it helps if the unit is not warm when initialising
- Broken camera module on the board; the cables and connectors are fragile and easy to damage.
- Frankly; some of the clones out there are very badly made.
  - I have several units, some are better than others, the ones I brought recently (late 2021) seem particularly cheap.
  - [rant] What happens is that manufacturers buy up 'out of spec' reject camera modules and esp moodules  from the back of the supply chain, then build modules using these and slip them out onto the market at a lowish price. There is nothing unusual in this, it has always happened, ever brought something 'cheap' that turned out to actually be 'defective', well that's capitalist economics working as intended, caveat emptor. And no, it's not just confined to China.
- Bad arduino core; running with a corrupted or outdated ESP arduino core in your build system will not help.

### Yes; I see errors like this occasionally
I understand that they are frustrating and a bit random, you have my sympathies and understanding; but I cannot provide any quick fixes.
- This is why I added the error catching, reporting and rebooting functionality to the sketch.
- _It always goes away after a while without me actually 'fixing' anything_.
  - The only board I had that didn't recover was because I had twisted the camera module on a board accidentally (broken ribbon cable). Fitting a new camera module fixed it, I have some wide-angle modules and spares.