easytarget/esp32-cam-webserver #206: Compile Error: esp32-cam-webserver-4.0.beta3.zip at line 531: error: 'CAMERA_GRAB_LATEST' was not declared in this scope

rolsch, 2022-01-03:
Compile Error: esp32-cam-webserver-4.0.beta3.zip at line 531

...\Documents\Arduino\esp32-cam-webserver\esp32-cam-webserver.ino: In function 'void setup()':
esp32-cam-webserver:531:12: error: 'struct camera_config_t' has no member named 'grab_mode'
     config.grab_mode = CAMERA_GRAB_LATEST;
            ^
esp32-cam-webserver:531:24: error: 'CAMERA_GRAB_LATEST' was not declared in this scope
     config.grab_mode = CAMERA_GRAB_LATEST;
                        ^
---
    config.pixel_format = PIXFORMAT_JPEG;
    **config.grab_mode = CAMERA_GRAB_LATEST;**
    // Pre-allocate large buffers
---

easytarget, 2022-01-05:
You need to upgrade to the latest ESP32 arduino core (currently v2.0.2).
- These errors indicate you have an older core, probably 1.x

bkbilly, 2022-01-07:
I also had the same issue.
I was using another Board Manager URL that the current version was not there.
I've changed to the one bellow and updated `esp32` and it works perfectly for me.
`https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json`

easytarget, 2022-01-08:
Thankyou;
I believe that ESP changed the URL for the boards manager quite some time ago; but not everybody got the memo, and the old URL is still active, but not updated.

I'm converting this to a discussion since I think this classifies as a 'Common Issue'