easytarget/esp32-cam-webserver #208: AI Thinker has no SPIFFS?

djusHa, 2022-01-06:
Hi,

please clarify, AI Thinker has no SPIFFS, right?

So i have to uncomment "#define NO_FS" in myconfig.h to get it run properly.

Is there any posibility save Setting to inserted SD Card?

Big thx!

djusHa, 2022-01-06:
Log, if "#define NO_FS" is commented (off):
```
====
esp32-cam-webserver: ESP32 camera server
Code Built: Jan  6 2022 @ 13:05:50
Base Release: 4.0.beta3
Camera init succeeded
OV2640 camera module detected
E (5886) SPIFFS: mount failed, -10025
ets Jul 29 2019 12:21:46

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12776
load:0x40080400,len:3032
entry 0x400805e4

====
```

djusHa, 2022-01-06:
Hardware info:
```
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: XXXXXXX
Stub is already running. No upload is necessary.
Manufacturer: 5e
Device: 4016
Detected flash size: 4MB
```

easytarget, 2022-01-07:
The AI-THINKER has SPIFFS.
- all four ESP-CAM boards (AI-THINKER) have working SPIFFS.
- Sorry; I'm guessing you have a defective board..
- ... or; you have the wrong board or partition scheme selected in the boards manager...
- ... or; you have a corrupted esp arduino core install.
The first thing to do is double-check the board selection and settings, and, if you know you have those correct, and you have the latest core, then it's almost certain your board is not up to spec.

How did you generate the hardware info dump?
- I'll happily run the same sketch/tool on my board so we can compare the results..

djusHa, 2022-01-07:
> How did you generate the hardware info dump?
You have to use esptool with chip_id or flash_id switch:
```esptool.exe --port COM4 chip_id```

> I'm guessing you have a defective board...
Yes, you right. Today it want start...

Esptool says:
```WARNING: Failed to communicate with the flash chip, read/write operations will fail.```

easytarget, 2022-01-08:
Thankyou,
For reference: On a working board I get:
```
$ ~/.arduino15/packages/esp32/tools/esptool_py/3.0.0/esptool.py --port /dev/ttyUSB0 flash_id
esptool.py v3.0-dev
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: ---
Uploading stub...
Running stub...
Stub running...
Manufacturer: 20
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...
```
and
```
$ ~/.arduino15/packages/esp32/tools/esptool_py/3.0.0/esptool.py --port /dev/ttyUSB0 chip_id
esptool.py v3.0-dev
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: ---
Uploading stub...
Running stub...
Stub running...
Warning: ESP32 has no Chip ID. Reading MAC instead.
MAC: ---
Hard resetting via RTS pin...
```

easytarget, 2022-01-08:
I'm closing and converting to a 'common issues' discussion thread to maximize visibility of this.