easytarget/esp32-cam-webserver #212: wifi Config enhancement

sekarpdkt, 2022-01-16:
I added an extension to allow user to configure various SSID list. If you are ok, will raise a PR. Basically, /wifi will bring the attached screenshot (page) and we can configure multiple SSIDs. Each SSID config is saved in a file. When SCAN happens, in addition to checking existing hardcoded list, it will also check for file availability. if file is found it is marked as available config. If best SSID is from file config, that file is used to connect to wifi. There are lots of changes in esp32-cam-webserver, app-httpd etc. Process:

1. Flash without any info.
2. Access Point is enabled
3. Connect to AP
4. Config Wifi info (by accessing <IP>/wifi (or click on new ICON added in main index page to access /wifi).
5. Save the config.
6. Reboot.
7. Cam will connect to new SSID. If IP address is given for that SSID, static IP is configured else DHCP IP is fetched.

If you feel it is aligned to your plan, will send a PR.
[image]

rdragonrydr, 2022-03-08:
_I_ would be quite interested in this feature... admittedly also for my own work. Do you have an AP/STA mode too?
I can't speak for the author, but I _think_ this was something they were planning eventually (at least according to the readme).

easytarget, 2022-03-09:
@sekarpdkt : I am very interested in this; but not for the upcoming release; I do not want to be making any extensive changes at this point.
A PR would be welcome of course, once 4.0 is out it can be rebased and I can go for a rapid v4.1 to implement it once tested and ready.
Somehow incorporating mesh networking would be nice, it is a good feature for this sort of device.

abratchik, 2022-12-03:
Is it possible to port this feature to the PR #280 ? This PR has major refactoring - all web pages previously stored as byte arrays are moved to SD card. Also all the Wi-Fi / connection code is Inca-splayed in classes, no more global / extern variables etc. 
Kindly have a look at this [fork](https://github.com/abratchik/esp32-cam-webserver)

gzoli1980, 2023-03-26:
Would you be interested in downloading the add-on?