easytarget/esp32-cam-webserver #214: "CRITICAL FAILURE: Camera sensor failed to initialise" on esp32-cam-webserver, but working on ESP32 CAM example sketch

cata-lin, 2022-01-29:
Hi, 
I struggle with a strange error while using an AI Thinker ESP32 CAM module. 
If using this AI Thinker ESP32 CAM with the ESP32 Camera example sketch, all works fine (there I selected "#define CAMERA_MODEL_AI_THINKER" and provided SSID and password before compiling). Additionally I checked the camera parts (I have 3 camera flex cable modules) using a TTGO ESP32 CAM module, they are all ok.

If using the esp32-cam-webserver, then I get this error:

E (7906) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
CRITICAL FAILURE: Camera sensor failed to initialise.
 
Before compiling I checked if the right ESP32 camera is selected in sketch (default it is selected), I provided SSID and password - same result. I checked that I selected PSRAM enabled boards, also partition scheme for OTA, I tried on board type "ESP32 Dev Module" as well as board type "AI Thinker ESP32 CAM"  

It is confusing that in the serial output is this part:
Camera init succeeded
OV2640 camera module detected

OTA is working fine, port is visible in Arduino IDE. I noticed on the serial output that SPIFFS is not there:
E (1311) SPIFFS: mount failed, -10025

Here is the serial output:

```
02:16:33.619 -> ets Jun  8 2016 00:22:57
02:16:33.619 -> 
02:16:33.619 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
02:16:33.619 -> configsip: 0, SPIWP:0xee
02:16:33.619 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
02:16:33.619 -> mode:DIO, clock div:1
02:16:33.652 -> load:0x3fff0030,len:1324
02:16:33.652 -> ho 0 tail 12 room 4
02:16:33.652 -> load:0x40078000,len:13508
02:16:33.652 -> load:0x40080400,len:3604
02:16:33.652 -> entry 0x400805f0
02:16:34.819 -> 
02:16:34.819 -> ====
02:16:34.819 -> esp32-cam-webserver: ESP32 camera server
02:16:34.852 -> Code Built: Jan 29 2022 @ 02:15:24
02:16:34.852 -> Base Release: 4.0.beta3
02:16:35.018 -> Camera init succeeded
02:16:35.018 -> OV2640 camera module detected
02:16:35.284 -> E (1311) SPIFFS: mount failed, -10025
02:16:36.184 -> ets Jun  8 2016 00:22:57
02:16:36.184 -> 
02:16:36.184 -> rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
02:16:36.184 -> configsip: 0, SPIWP:0xee
02:16:36.184 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
02:16:36.184 -> mode:DIO, clock div:1
02:16:36.184 -> load:0x3fff0030,len:1324
02:16:36.184 -> ho 0 tail 12 room 4
02:16:36.217 -> load:0x40078000,len:13508
02:16:36.217 -> load:0x40080400,len:3604
02:16:36.217 -> entry 0x400805f0
02:16:37.382 -> 
02:16:37.382 -> ====
02:16:37.382 -> esp32-cam-webserver: ESP32 camera server
02:16:37.382 -> Code Built: Jan 29 2022 @ 02:15:24
02:16:37.428 -> Base Release: 4.0.beta3
02:16:44.437 -> E (7906) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
02:16:44.538 -> 
02:16:44.538 -> 
02:16:44.538 -> CRITICAL FAILURE: Camera sensor failed to initialise.
02:16:44.538 -> 
02:16:44.538 -> A full (hard, power off/on) reboot will probably be needed to recover from this.
02:16:44.538 -> Meanwhile; this unit will reboot in 1 minute since these errors sometime clear automatically
02:16:44.538 -> Lamp: 0%, pwm = 0
02:16:45.236 -> Starting WiFi
02:16:45.236 -> Known external SSIDs:  'ABCDE'
02:16:45.236 -> MAC address: E8:68:CC:CC:CC:CC
02:16:45.269 -> Scanning local Wifi Networks
02:16:51.265 -> 20 networks found
02:16:51.265 ->   1 : [11:11:11:11:11:11] ABCDE (-50)  -  Known!
02:16:51.265 ->   2 : [11:11:D5:8B:35:B1] JustUs (-52)
02:16:51.265 ->   3 : [00:11:67:09:14:CB] TP-Link_14CC (-62)
02:16:51.265 ->   4 : [D8:32:14:2E:03:91] Tenda_2E0390 (-63)
02:16:51.297 ->   5 : [70:4F:11:EA:CD:B6] TP-Link_BCB6 (-66)
02:16:51.297 ->   6 : [80:E1:BF:47:31:5C] DIGI-y26U (-68)
02:16:51.297 ->   7 : [F4:F2:6D:E7:B5:E6] Uaifai Reloaded (-69)
02:16:51.297 ->   8 : [C8:3A:35:01:3A:70] Ramona (-70)
02:16:51.297 ->   9 : [20:F1:7C:32:9F:94] TOSHI (-70)
02:16:51.297 ->  10 : [34:97:F6:A5:E6:1C] FLY79 (-73)
02:16:51.297 ->  11 : [88:44:77:0F:CE:90] Telekom-011778 (-73)
02:16:51.297 ->  12 : [88:44:77:0F:CE:93] Telekom_FON (-73)
02:16:51.330 ->  13 : [38:6B:1C:28:F8:E9] WIFI AC1900 (-74)
02:16:51.330 ->  14 : [80:E1:BF:48:A2:C8] YUNoLikeWires? (-82)
02:16:51.330 ->  15 : [58:D9:D5:DF:B7:51] Tenda_DFB750 (-84)
02:16:51.330 ->  16 : [BC:C0:0F:6B:D1:D0] MauHome (-88)
02:16:51.330 ->  17 : [80:E1:BF:47:C0:88] DIGI-Mamba (-93)
02:16:51.330 ->  18 : [18:0F:76:38:F3:F0] WIFI n/ac (-93)
02:16:51.330 ->  19 : [AC:84:C6:03:1C:D5] YUNoLikeWires? (-95)
02:16:51.330 ->  20 : [00:5F:67:09:FB:23] Caraibe (-98)
02:16:51.330 -> Connecting to Wifi Network 0: [DC:CD:11:AA:2D:94] ABCDE 
02:16:51.863 -> ..............Client connection succeeded
02:16:58.356 -> IP address: xxx.168.0.100
02:16:58.356 -> Netmask   : 255.255.255.0
02:16:58.356 -> Gateway   : xxx.168.0.101
02:16:58.356 -> Setting httpURL
02:17:00.354 -> Setting up OTA
02:17:00.354 -> 
02:17:00.354 -> No OTA password has been set! (insecure)
02:17:00.354 -> 
02:17:00.354 -> Time functions disabled
02:17:00.354 -> Starting web server on port: '80'
02:17:00.387 -> Starting stream server on port: '81'
02:17:00.387 -> 
02:17:00.387 -> Camera unavailable due to initialisation errors.
02:17:00.387 -> 
02:17:01.453 -> 
02:17:01.453 -> This is the 4.0 alpha
02:17:01.453 ->  - Face detection has been removed!
02:17:44.539 -> E (82666) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
02:17:44.539 -> E (82666) task_wdt:  - loopTask (CPU 1)
02:17:44.539 -> E (82666) task_wdt: Tasks currently running:
02:17:44.539 -> E (82666) task_wdt: CPU 0: IDLE
02:17:44.572 -> E (82666) task_wdt: CPU 1: IDLE
02:17:44.572 -> E (82666) task_wdt: Aborting.
02:17:44.572 -> 
02:17:44.572 -> abort() was called at PC 0x400e6ee5 on core 0
02:17:44.572 -> 
02:17:44.572 -> 
02:17:44.572 -> Backtrace:0x40083a95:0x3ffc02dc |<-CORRUPTED
02:17:44.572 -> 
02:17:44.572 -> 
02:17:44.572 -> 
02:17:44.572 -> 
02:17:44.572 -> ELF file SHA256: 0000000000000000
02:17:44.572 -> 
02:17:44.572 -> Rebooting...
```

Here is my configuration: Ubuntu 20.04, Arduino Genuino 1.8.19, esp32 by espressif 2.0.2
Any hint is appreciated, thank you :)
Catalin

reilnuud, 2022-01-30:
I fixed this by pulling the latest *stable* espressif32 instead of the latest dev version.

(I'm using platformio so it was changing line 15 in platformio.ini to

`platform = espressif32`

nassauwinter, 2022-02-09:
Hi, it seems I got the same error:
```23:31:15.272 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
23:31:15.318 -> configsip: 0, SPIWP:0xee
23:31:15.318 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
23:31:15.318 -> mode:DIO, clock div:1
23:31:15.318 -> load:0x3fff0030,len:1324
23:31:15.318 -> ho 0 tail 12 room 4
23:31:15.318 -> load:0x40078000,len:13508
23:31:15.318 -> load:0x40080400,len:3604
23:31:15.318 -> entry 0x400805f0
23:31:16.068 -> 
23:31:16.068 -> ====
23:31:16.068 -> esp32-cam-webserver: ESP32 camera server
23:31:16.068 -> Code Built: Feb  8 2022 @ 20:49:32
23:31:16.115 -> Base Release: 4.0.beta3
23:31:16.256 -> Camera init succeeded
23:31:16.256 -> OV2640 camera module detected
23:31:16.537 -> E (891) SPIFFS: mount failed, -10025
```
Win10, Arduino 1.8.19, esp32 2.0.2, Some generic ESP32-CAM module with ai thinker camera pins.
[image]

The hardware tested with espressif example and operates normally. 
Tried several releases but spiffs init failed:
[image]

TungstenE2, 2022-02-13:
but this will not work via OTA. At least step 2, as ESP lost WIFI info after step1, right?

sptim, 2022-02-19:
If the SPIFFS is initialized before the camera everything works like expected
[diff.txt](https://github.com/easytarget/esp32-cam-webserver/files/8102796/diff.txt)
.

easytarget, 2022-03-08:
Ok, so there is something interesting here; as @sptim identifies; things work if the SPIFFS initialisation is done before the camera; and I'm going to re-order the startup code as he suggests.

What is weird is that the example SPIFFS sketch does not do anything differently from my sketch, they both start the filesystem with the correct 'format if needed' setting. The important difference, I think, is that this sketch has the camera module initialised and running, but the SPIFFS example does not. 
eg: The root cause appears to be bus/camera/spiffs conflicts; looking at this and the issues I linked to above it seems to me that the reported SPIFFS and camera failures are related; lowering the xclk bus frequency is one of the ways they can be resolved.

The 4.0 RC has several changes to xclk handling; it is set low by default during startup now; and the user can increase it in the UI and save that to the preferences.
I'm hoping that those changes; and starting SPIFFS before the camera, will solve this.

easytarget, 2022-03-08:
xclk stuff is in de01547

closed 2022-03-08 by easytarget