easytarget/esp32-cam-webserver #216: Spiffs: No filesystem found

TungstenE2, 2022-02-09:
Just tried to update to the latest version.

Cam modul fails to load.

[image]

Board settings are according to docu.

[image]

[image]

What is wrong?

nassauwinter, 2022-02-09:
@TungstenE2 It seems to be related with #214. Got "SPIFFS: no filesystem found" error too.

TungstenE2, 2022-02-12:
@easytarget could you please look into this? thx a lot

TungstenE2, 2022-02-13:
this did the trick also for me:

The two steps:

    - flash SPIFFS_Test.ino from esp32 sketch example library
    - flash esp32-cam-webserver.ino

Cam modul is loading and stream and still is working again. Only dump page is not loading...

https://github.com/easytarget/esp32-cam-webserver/issues/214#issuecomment-1038100057

easytarget, 2022-03-08:
Rolling this and all related into a single issue: #214

closed 2022-03-08 by easytarget