easytarget/esp32-cam-webserver #217: Stream available Webserver not

gfucka, 2022-02-11:
After hours the cam is on the Webserver Is not available. I can whatch camera stream at Port 81 but the configuration page on Port 80 Is not visibile with ERR_EMPTY_RESPONSE

easytarget, 2022-03-05:
Thankyou for the report, are you streaming the whole time?
On my test cameras I never see issues like this; I can always access the settings and status screens even after weeks of uptime. But I do not stream continually from them; I'm wondering if this is a factor?

gfucka, 2022-03-05:
Yes I'm using It with motioneye, so It stream continually.

easytarget, 2022-03-08:
For reference.. I'm currently streaming from my workshop cam to my server. The cam is permanent and already has 55+ days of uptime. After a day the stream is still running and so are the main UI and dump pages.. 
I'll leave it running for a bit, but, basically, I'm not seeing any issues.

easytarget, 2022-03-14:
I cannot reproduce this; I had a stream runnign for 2+days to a browser window on my media machine, and could still connect to the main controls and dump page whenever I tried. Without being able to reproduce I cannot investigate further, closing.

closed 2022-03-14 by easytarget

gfucka, 2022-10-11:
Ok thanks for testing.