easytarget/esp32-cam-webserver #227: Help with IDE

OwenD30, 2022-03-08:
Hello, I am trying to compile but when I follow the directions and open the "esp32-cam-websever" folder and import the "esp32-cam-webserver.ino", only one tab shows up. No configuration.h or anything, just the webserver tab. I know this issue is totally my fault because I don't have much experience but could you tell me what I did wrong? Thanks!

easytarget, 2022-03-08:
I assume you are using the Arduino IDE?
In the Arduino IDE go to `file` -> `Open...`, then browse to the `.ino` file and open that.
This **should** open the entire sketch folder in tabs, this is how it works on my system. There is no 'import' button in the IDE, so I'm a bit confused to be honest.

OwenD30, 2022-03-08:
By "import" I meant open, sorry for the confudion. I will send a screen recording later today when I get the time. Maybe I'm opening the wrong file or something.

OwenD30, 2022-03-08:
I also tried it on the chrome version of Arduino IDE and the same thing happened. I attached a screen recording of the only tab that shows up when I open "esp32-cam-webserver.ino"

https://user-images.githubusercontent.com/77121604/157240496-62dffd67-7d52-44a2-b8a9-181f740b5dff.mp4

.

easytarget, 2022-03-08:
Humm. I've never used, or even tried, the web IDE with this sketch.
- All the documentation assumes you are using the native IDE, and that is the only IDE I support.
Maybe the web IDE has a 'open sketch' dialog that lets you open an entire folder?

My apologies, but I don't have space and time to test, debug and support this. Your best bet is to go to the Arduino Web IDE support forums and ask 
1) does the Web IDE support the esp32 (a 10 second google suggests not)
1) If so, how to open the whole sketch folder with it

easytarget, 2022-03-08:
Ah.
Make sure your files are unpacked to a folder called `esp32-cam-webserver`. Remove any version part of the folder name. 
For the ide to open the whole sketch the folder must have exactly the same name as the.`.ino` file, but without the `.ino` extension.

OwenD30, 2022-03-08:
I did that and added it to my arduino library, but when I try to open it it doesn't let me open the whole folder, only the .ino. I'm sorry, I'm sure it's something very simple but I just don't know :-)

https://user-images.githubusercontent.com/77121604/157334557-2d12927d-a1f7-435e-a2d0-e0c2d48ce1b2.mp4

easytarget, 2022-03-09:
This is not a library, do not put it in the libraries folder. Put it in the Arduino folder.
If that does not work then please take this up in the Arduino forums; I cannot provide basic IDE use hand-holding here.

closed 2022-03-09 by easytarget

OwenD30, 2022-03-09:
Another issue, this time with the sketch itself. Got this when I tried to verify:
[image]

easytarget, 2022-03-09:
That one is easy; you need to upgrade the Arduino IDE core in the boards manager; I only support the latest (2./0.2) version with 
the latest versions of the sketch.

easytarget, 2022-03-09:
#224