easytarget/esp32-cam-webserver #247: Crop or zoom image/stream

adolfocartagena, 2022-04-17:
Hello everybody!

Would it be possible to crop the image to a certain area? I have a camera in the peephole and only 70% of the image is useful
[image]
, the rest is black. Thank you

therealfreezoid, 2022-09-30:
I'd love to see this feature in here! In my current project both sides of the video footage are unnecessary and distracting. Would be nice to cut it out or at least cover with black bars. 
Just leaving a comment in case it will help convince someone for implementation! :)

easytarget, 2024-12-09:
I am leaving this issue open for reference; although I am archiving the project