easytarget/esp32-cam-webserver #250: Can't start stream from control page from outside gateway

80stech, 2022-04-25:
It seems that the control page trys to start the stream on port 81 using the local IP and so fails to work from outside of the gateway. Streaming works fine when I access it using the gateway URL and (port forwarding) port 81.  This is all new to me so I apologize if this is a silly problem. :)

80stech, 2022-04-25:
Ok, I think this might be that I need to have the host address listed first in the list of the connections. I'll give that a try :)

easytarget, 2022-05-05:
Note that I dont really support streaming via routers/gateways etc, in part because of #1 (which may be resolved soon) but also because there are too many variables and different ways it can be done and configured. 

The long-and-short of it is that you must expose both ports (control and streaming, by default `80` and `81`) and adjust the default UI ports to match the exposed ones in the config.

closed 2022-05-05 by easytarget