easytarget/esp32-cam-webserver #252: Test with Arduino IDE Core 2.0.3

easytarget, 2022-05-05:
https://github.com/espressif/arduino-esp32/releases/tag/2.0.3

oerkel47, 2022-05-07:
I saw the reply about 2.0.3 in that other issue and have been running this version with PlatformIO since. Appears to be working fine.

easytarget, 2022-05-07:
Looking good so far.

easytarget, 2022-05-08:
No obvious problems encountered, I havnt tested /every/ scenario and config combo but the basics all look good. 

I've committed a change to how `grab_mode` is handled at startup, and removed some redundant psram related code as a result of investigating this; 4c9ab6e

Good news is that the 2.0.3 board selection for `AI-Thinker ESP-32 CAM` now allows you to select the partition scheme (https://github.com/espressif/arduino-esp32/issues/6114), I can document this in the README.

closed 2022-05-08 by easytarget