easytarget/esp32-cam-webserver #260: Android display issues - Stream failed, code = 45062 : ESP_ERR_HTTPD_RESP_SEND#

yavinmaster, 2022-06-25:
I've tried 3 modules now and see the same problem with all of them.  Connecting using a Windows 10 Laptop and Chrome and all works well.  If I connect using either an Android tablet or phone then the stream works for about 15 seconds then stops.  I've played with several settings but nothing seems to help.  I've tried chrome and firefox on android.  I've also tried "desktop mode".  I've tried direct 5V supplies and USB power so I can keep an eye on the serial coms - this potentially gives the only clue as to what is going wrong where I sometimes receive the error - "Stream failed, code = 45062 : ESP_ERR_HTTPD_RESP_SEND#"
To be clear, I have the same issue with the standard Arduino demo for the esp32-cam.

80stech, 2022-06-25:
My guess would still be power supply. I found I need a 5v 2amp power supply to have reliable streaming. Putting a 30 - 100 uf cap across the supply pins helps a lot too with a 1 amp supply but still not reliable streaming.

dilmash, 2022-08-22:
Same issue here but only on ESP32 WROOM chips. I have then used the ESP32S chip and have no problems.

yavinmaster, 2022-10-11:
Thanks for the suggestion, but with a 100uF capacitor fitted and connecting
to a bench top power supply set to 5V and 2A limit, I see exactly the same
issue.

derrapf, 2023-07-06:
Same for me. Wo appreaciate a solution currently to unstable to really use it in home automation

uvish, 2023-08-01:
same issue here , getting this error every few mins. This and the random freezing after few days is stopping me from using this as a 24/7 cam.

easytarget, 2024-12-06:
Sorry; I am not doing any further development on this project. it is old code and somewhat obsolete. I am currently working on retiring it; I will try to link to a more actively updated fork when I do.

closed 2024-12-06 by easytarget