easytarget/esp32-cam-webserver #266: Feature Request: Auto reboot

TungstenE2, 2022-08-20:
Hi all,

I have a cam at my door, which is not responding after some days in operation. A second cam with same fw is working fine for several weeks.

So I was wondering if there is an option to reboot the cam very night at a specific time?

Either by code enhacement or by external http comand.

thx

TungstenE2, 2022-11-02:
push ;-)

abratchik, 2022-12-26:
@TungstenE2 I think you can use the `http://<your-ip:your-port>/cmdout?var=reboot&val=1` command. You can also trigger it by a schedule with help of any external tool.

easytarget, 2024-12-06:
Sorry; I am not doing any further development on this project. it is old code and somewhat obsolete. I am currently working on retiring it; I will try to link to a more actively updated fork when I do.

closed 2024-12-06 by easytarget