easytarget/esp32-cam-webserver #269: 'struct camera_config_t' has no member named 'fb_location'

Terbinm, 2022-09-16:
I can't pass the compile the code with error:
'struct camera_config_t' has no member named 'fb_location'

d3viant, 2022-10-19:
Solved by upgrading to Arduino IDE 2.0.

Muplex, 2022-10-29:
Same error here

Muplex, 2022-11-01:
I try in another PC 
fresh installation windows, arduino IDE and so on... same resoult

`C:\Users\nicol\OneDrive\Documenti\Arduino\Esperimenti e Progetti\esp32-cam-webserver\esp32-cam-webserver.ino: In function 'void StartCamera()':
C:\Users\nicol\OneDrive\Documenti\Arduino\Esperimenti e Progetti\esp32-cam-webserver\esp32-cam-webserver.ino:341:12: error: 'struct camera_config_t' has no member named 'fb_location'
     config.fb_location = CAMERA_FB_IN_PSRAM;
            ^
C:\Users\nicol\OneDrive\Documenti\Arduino\Esperimenti e Progetti\esp32-cam-webserver\esp32-cam-webserver.ino:341:26: error: 'CAMERA_FB_IN_PSRAM' was not declared in this scope
     config.fb_location = CAMERA_FB_IN_PSRAM;
                          ^
C:\Users\nicol\OneDrive\Documenti\Arduino\Esperimenti e Progetti\esp32-cam-webserver\esp32-cam-webserver.ino:343:12: error: 'struct camera_config_t' has no member named 'grab_mode'
     config.grab_mode = CAMERA_GRAB_LATEST;
            ^
C:\Users\nicol\OneDrive\Documenti\Arduino\Esperimenti e Progetti\esp32-cam-webserver\esp32-cam-webserver.ino:343:24: error: 'CAMERA_GRAB_LATEST' was not declared in this scope
     config.grab_mode = CAMERA_GRAB_LATEST;
                        ^
Multiple libraries were found for "WiFi.h"
  Used: C:\Users\nicol\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi
  Not used: C:\Users\nicol\OneDrive\Documenti\Arduino\libraries\WiFiEspAT
Multiple libraries were found for "ArduinoOTA.h"
  Used: C:\Users\nicol\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\ArduinoOTA
  Not used: C:\Users\nicol\OneDrive\Documenti\Arduino\libraries\ArduinoOTA
exit status 1

Compilation error: 'struct camera_config_t' has no member named 'fb_location'`

Muplex, 2022-11-19:
Where I can find sketch lastest version?

Muplex, 2022-11-19:
> IIRC, that is the error you get with the old sketch using the latest esp32 core. The fix was to revert the core back to 1.0.6 but I think the latest version of the sketch does not have that error.

Where I can find sketch lastest version?

Muplex, 2022-11-19:
> ESP32 boards should be 2.0.5 for Ver 4.

I thin kyou mean 1.0.5 esp32 boards (lastest is 1.0.6)

I try right now to downgrade Esp32 board to 1.0.5 but same resoult same error same lines  :(

any ideas?

Muplex, 2022-11-19:
[image]

Sorry i'm missing something.

additional board URL is corect ?

`http://arduino.esp8266.com/stable/package_esp8266com_index.json,https://dl.espressif.com/dl/package_esp32_index.json`

Muplex, 2022-11-19:
Thank you I solved, the problem was the additional board URL's 
=)

JeanPoll, 2023-01-06:
Hmm same issue here, URS are fine, don't know how to fix.

briodan, 2023-07-22:
running into the same issue

[image]

additional board URL's

http://arduino.esp8266.com/stable/package_esp8266com_index.json,
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json

installed version
[image]

any clue on what else to try?

briodan, 2023-07-24:
on a whim i tried the windows store version of the arduino ide which installed a different version

[image]

which compiled correctly, so i guess an older/different version is what was needed for me,
All else is the same.

easytarget, 2024-12-06:
Some good info here; I'm also not closing this yet since I want to capture this as I archive the project