easytarget/esp32-cam-webserver #276: src/parsebytes.h: No such file or directory

sbx59, 2022-11-16:
hello, 
I just followed the tutorial, but I have two errors. Can you help me ?

Plusieurs bibliothèque trouvées pour "WiFi.h"
esp32-cam-webserver:8:10: fatal error: src/parsebytes.h: No such file or directory
Utilisé : C:\Users\USER\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.5\libraries\WiFi
 #include "src/parsebytes.h"
Non utilisé : C:\Program Files (x86)\Arduino\libraries\WiFi
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
src/parsebytes.h: No such file or directory

[image]

MavricFloquet, 2022-11-18:
Hello, i have the same probleme do you have any issu ? THX

sbx59, 2022-11-19:
Good for me now .

sore-ron-git, 2023-02-19:
Issue occurs when you rename the zip file from esp32-cam-webserver-master.zip to esp32-cam-webserver.zip.

When you extract this file it creates esp32-cam-webserver-master folder.

You need to rename this folder as well to esp32-cam-webserver

easytarget, 2024-12-06:
Sorry; I am not doing any further development on this project. it is old code and somewhat obsolete. I am currently working on retiring it; I will try to link to a more actively updated fork when I do.

closed 2024-12-06 by easytarget