easytarget/esp32-cam-webserver #287: Getting No PSRAM found on first boot

Drizzt321, 2023-02-03:
So I got [this](https://smile.amazon.com/gp/product/B08GKLWCB7) esp32-cam, and when using your sketch I get `No PSRAM found; camera cannot be initialised: Please check the board config for your module.` as an error on first boot. Screenshot and full output is below.

I did ensure in Tools->PSRAM -> Enabled is set, using latest Arduino (2.0.3) and esp32 boards (2.0.6) is installed.

[image]

```
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13836
load:0x40080400,len:3608
entry 0x400805f0
E (242) psram: PSRAM ID read error: 0xffffffff

====

esp32-cam-webserver: ESP32 camera server
Code Built: Feb  3 2023 @ 11:15:57
Base Release: 4.0

Fatal Error; Halting
No PSRAM found; camera cannot be initialised: Please check the board config for your module.
```

Drizzt321, 2023-02-03:
Instructions in the README says to use the generic `ESP32 Dev Module`

Drizzt321, 2023-02-03:
Set the board to `AI Thinker ESP32-CAM`, same issue, `No PSRAM found`. 

Note, this is NOT an AI-Thinker branded board, see Amazon link I have above, is what I'm trying to use. Wonder if this particular module, ESP32-S, just doesn't have any PSRAM.

Drizzt321, 2023-02-03:
What were your compile settings? I didn't "mess around", I followed the README on the project. If that's wrong, then please tell me what the correct settings are.

Ropese, 2023-04-17:
Good morning
I also have the psram problem I tried 3 esp32-s only one goes the others psram error I changed the camera modules but still the same.
how can it be that 1 goes and the others don't?
i tried with other skretch and other modules work too but not with this any idea?
Thank you

Emstar, 2023-10-04:
I have the same kind of ESP32-S board and had the same PSRAM problem. 

I found the solution in this post: https://www.esp32.com/viewtopic.php?t=31247#p107434

In `C:\Users\USERNAME\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.13\cores\esp32\esp32-hal-psram.c` rig the PSRAM test:

```
//#if CONFIG_SPIRAM_SUPPORT || CONFIG_SPIRAM
#if 1
```

In `esp32-cam-webserver.ino`, in `setup()`, add this after the block of startup `serial.print()` statements near line 650.

```
    if (psramInit()) {
      Serial.print("PSRAM OK. Size: ");
      Serial.println(ESP.getFreePsram());
      }
```

In `myconfig.h` continue to use the default `AI_THINKER` definition. 

While this worked there is probably a better way to do it. This is my first project, I don't know this environment or hardware at all. Modifying a core library file sucks and seems like it would break compatibility with other boards. 

I also noticed that PSRAM did not seem to initialize properly on every bootup. Sometimes you have to push the reset button to get the board to come up. 

If anyone can suggest the right way to implement this support without messing with `esp32-hal-psram.c` I would love to know.

easytarget, 2024-12-06:
Sorry; I am not doing any further development on this project. it is old code and somewhat obsolete. I am currently working on retiring it; I will try to link to a more actively updated fork when I do.

closed 2024-12-06 by easytarget