easytarget/esp32-cam-webserver #294: cam hangs after 2-3 days

TungstenE2, 2023-08-06:
About avery 3-4 days the cam hangs and can only be restarted by power disconnect.

Even Reboot by URL every 1,5 hrs does not solve this.

Any ideas?

Running on latest version.

adameus03, 2023-08-30:
@TungstenE2 I have 4 AI-Thinker ESP32-CAM's. I'm not running the firmware from this repo, but for my surveillance project, 1 out of 4 of the cameras fails after 10 minutes. By failure I mean the frame buffer pointer returned by the esp_camera_fb_get() function from the "esp_camera.h" library becomes NULL. Software reset or even pressing the ESP32-CAM built-in reset button doesn't change this behaviour. The only solution is disconnect the power supply and reconnecting it again. The other 3 boards are working fine for weeks without this issue.
With that said, it is possible you have a corrupted board & need to purchase another AI-Thinker ESP32-CAM

TungstenE2, 2023-09-20:
I changed the board and cam is now stable for over a week, Only difference is the wifi antenna is now external.

closed 2024-12-06 by easytarget

TungstenE2, 2024-12-06:
sad to hear, but fully understand

easytarget, 2024-12-06:
@TungstenE2 ; I'd like to specifically thank you for all the help you gave here!

I will do  proper wrap up. I am hoping to begin the README with  a clear 'my code works with IDE2.0.2 and here is how you do that' section; then discuss oalternatives, then retire the rest and make the repo archived.

I should have done this some time ago but a) I'm a coward and b) I had moved on to shinier things ;-)