easytarget/esp32-cam-webserver #302: Error while compiling with Arduino IDE 1.8.19

SchneHa, 2023-11-07:
app_httpd.cpp:152:19: error: format '%i' expects argument of type 'int', but argument 4 has type 'long unsigned int' [-Werror=format=]
     Serial.printf("CPU Freq: %i MHz, Xclk Freq: %i MHz\r\n", ESP.getCpuFreqMHz(), xclk);
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                       ~~~~

What board definition and library versions have to be present?

SchneHa, 2023-11-07:
Board definition must be at least 2.0.2. Don't know if newer definitions work.

easytarget, 2024-12-06:
No; none of the newer definitions work.
Sorry; I am not doing any further development on this project. it is old code and somewhat obsolete. I am currently working on retiring it; I will try to link to a more actively updated fork when I do.

closed 2024-12-06 by easytarget