easytarget/esp32-cam-webserver #306: Error when compiling for Xiao ESP32S3

dsssssssss9, 2023-12-10:
HI
Hope someone can help

I have just got a SEEED Xiao ESP32S3 sensoe board + Camera

When i try to compile this i get the following errors....

> /home/jkd/.arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /tmp/arduino/sketches/9371D3A03917BEC5430B1019A3C12EAB/sketch/app_httpd.cpp.o:(.literal._Z10serialDumpv+0xf4): undefined reference to `temprature_sens_read'
/home/jkd/.arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /tmp/arduino/sketches/9371D3A03917BEC5430B1019A3C12EAB/sketch/app_httpd.cpp.o: in function `serialDump()':
/home/jkd/Arduino V2 - 2022/SketchBook/esp32-cam-webserver-/esp32-cam-webserver/app_httpd.cpp:147: undefined reference to `temprature_sens_read'
/home/jkd/.arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /home/jkd/Arduino V2 - 2022/SketchBook/esp32-cam-webserver-/esp32-cam-webserver/app_httpd.cpp:148: undefined reference to `temprature_sens_read'
/home/jkd/.arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /tmp/arduino/sketches/9371D3A03917BEC5430B1019A3C12EAB/sketch/app_httpd.cpp.o: in function `dump_handler(httpd_req*)':
/home/jkd/Arduino V2 - 2022/SketchBook/esp32-cam-webserver-/esp32-cam-webserver/app_httpd.cpp:596: undefined reference to `temprature_sens_read'
/home/jkd/.arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /home/jkd/Arduino V2 - 2022/SketchBook/esp32-cam-webserver-/esp32-cam-webserver/app_httpd.cpp:597: undefined reference to `temprature_sens_read'
collect2: error: ld returned 1 exit status
exit status 1
Compilation error: exit status 1

Anyone know how to solve this?

Thanks in advance

ali86x, 2024-02-03:
Same problem here.
The original code for ESP32 Webscam Server works, but this version gets this error

easytarget, 2024-12-06:
Newer ESP chips need newer ESP idf frameworks; but my code is not compatible with these. I guess you are hitting one of the differences from the S3 to the original.

Sorry; I am not doing any further development on this. it is old code and somewhat obsolete. I am currently working on retiring it; I will try to link to a more actively updated fork when I do.

closed 2024-12-06 by easytarget