easytarget/esp32-cam-webserver #311: how to get working with ov2640

cozmo14047, 2024-02-25:
hi there how can i get this working with the ov2640 camera

cozmo14047, 2024-02-29:
I figured it out you have to change the pin definition for the ai thinker board if you use alixpress clones

tobiasisenberg, 2024-08-29:
But how specifically? Can you provide details or a link to the respective resource?

cozmo14047, 2024-08-30:
> But how specifically? Can you provide details or a link to the respective resource?
Jeez, it's been a while since I got this working—I completely forgot about this thread, lol. Do you have Discord? It'd be easier to help you there. There wasn't a single resource; it involved a lot of digging through schematics, source code, etc.

tobiasisenberg, 2024-08-30:
My hunch now is that my own camera is somehow broken, as it used to work before and now does not work with 2 ESP32-CAM modules in which other cameras work. Nonetheless, I found several different pinouts online, but mostly for other modules with a USB-C connector, not the plain ESP32-CAM modules w/o USB connector. But if you still have your code somewhere it may be worth it to post your own changed pin assignment, so that others can compare it to theirs.

cozmo14047, 2024-09-08:
> My hunch now is that my own camera is somehow broken, as it used to work before and now does not work with 2 ESP32-CAM modules in which other cameras work. Nonetheless, I found several different [...]

It could be broken, but that's unlikely unless the ribbon cable is damaged. If I find it, I'll upload it somewhere, though it's probably buried deep in my hard drive. So, do you only have the camera module and not the programming board? If that's the case, you'll need an Arduino or another AVR programmer to program them, I believe.

tobiasisenberg, 2024-09-09:
I do have the programming board, And a new camera module is on the way. I am pretty confident it is the camera.

cozmo14047, 2024-09-09:
> I do have the programming board, And a new camera module is on the way. I am pretty confident it is the camera.

Ok let me know if it still doesnt work and ill find the code

tobiasisenberg, 2024-09-11:
It was the camera: it works fine with a new one. So all is good now on my side. Nonetheless, if you come across the changed config it would be nice to have a link here or the actual pin numbers that need to change.

cozmo14047, 2024-09-11:
> It was the camera: it works fine with a new one. So all is good now on my side. Nonetheless, if you come across the changed config it would be nice to have a link here or the actual pin numbers that need to change.

Thats good that its working now, is there a chance it could be the pins on the other camera are just diffrent? Yeah ive been looking im pretty sure i know know where the config is if so ill make a pull request here to add the option

easytarget, 2024-12-06:
Sorry; I am not doing any further development on this. it is old code and somewhat obsolete. I am currently working on retiring it; I will try to link to a more actively updated fork when I do.

closed 2024-12-06 by easytarget