easytarget/esp32-cam-webserver #312: Streaming on Andorid Phone not possible

JoJos1220, 2024-02-26:
As descriped in the title, it is not possible in the simple/full mode by activating start stream to get a stream view. It works in Browser (Firefox) perfectly as excepted but the stream does not work on Android.

The Get still button gives a screen but no moving stream.

JoJos1220, 2024-02-26:
As a Breakpoint:

the issue occurs if the URL_HOSTNAME & the MCNS_NAME was activated and setted up exactly the same string. Maybe this is a helpful information

easytarget, 2024-12-06:
Sorry; I am not doing any further development on this. it is old code and somewhat obsolete. I am currently working on retiring it; I will try to link to a more actively updated fork when I do.

closed 2024-12-06 by easytarget

easytarget, 2024-12-09:
It is, however, closed for this repository. 
In any case this sounds like a issue with the browser on your phone, not with the cam server itself.