easytarget/esp32-cam-webserver #319: Watchdog API changes in  arduino-esp32 version 3.x.x

TylerWhittaker, 2024-07-19:
There are some breaking API changes related to watchdog functionality in the recently released 3.x.x branch of [arduino-esp32](https://github.com/espressif/arduino-esp32).  

> The Interrupt Watchdog API (functions/types/macros prefixed with esp_int_wdt_) has been made into a private API. Thus, the previous include path #include "esp_int_wdt.h" has been removed. If users still require usage of the Interrupt Watchdog API (though this is not recommended), it can be included via #include "esp_private/esp_int_wdt.h".

So you're probably going to need to update `esp32-cam-webserver.ino` and `app_httpd.cpp` with something like: 
```
#if ESP_ARDUINO_VERSION_MAJOR >= 3
    #include "esp_private/esp_int_wdt.h"
#else
    #include "esp_int_wdt.h"
#endif
```

> The API for esp_task_wdt_init() has changed as follows:
>   -   Configuration is now passed as a configuration structure.
>   -   The function will now handle subscribing of the idle tasks if configured to do so.

- https://docs.espressif.com/projects/esp-idf/en/release-v5.1/esp32/migration-guides/release-5.x/5.0/system.html
- https://github.com/espressif/arduino-esp32/blob/master/docs/en/migration_guides/2.x_to_3.0.rst#migration-from-2x-to-30

dsssssssss9, 2024-10-10:
has there been any progress on this please??

I am still unable to compile

easytarget, 2024-12-06:
Sorry; I am not doing any further development on this. it is old code and somewhat obsolete. I am currently working on retiring it; I will try to link to a more actively updated fork when I do.

closed 2024-12-06 by easytarget